STMicroelectronics/STM32CubeWB #102: Failed to erase sectors when downloading the program for the first time

huangliguang, 2024-09-09:
test STM32Cube_FW_WB_V1.19.0\Projects\P-NUCLEO-WB55.Nucleo\Examples\FLASH\FLASH_EraseProgram
Failed to erase sectors when downloading the program for the first time
After the debugger is reset, the erase is successful again, only the first erase of the downloaded program will fail

TOUNSTM, 2024-09-12:
Hello @huangliguang,

First, we would like to thank you for your report.

We attempted to reproduce the issue you described using `IAR 9.30.1` and `STM32CubeIDE v1.14.0` compilers with `STM32Cube_FW_WB_V1.19.0` and `STM32Cube_FW_WB_V1.20.0` packages, but we were unable to replicate the problem.

Could you please provide more details about your setup, including Any modifications you made to the example project or any additional information that might help us understand the context better?

Thank you for your cooperation.
kind regards,

huangliguang, 2024-09-13:
> Hello @huangliguang,
>
> First, we would like to thank you for your report.
>
> We attempted to reproduce the issue you described using `IAR 9.30.1` and `STM32CubeIDE v1.14.0` compilers with [...]

download and enter debug mode，erase sectors will be fail

TOUNSTM, 2024-09-13:
Hello @huangliguang,

The `STM32WB` series microcontrollers support only page erase operations, not sector erase operations. The example provided in the `STM32Cube_FW_WB_V1.19.0` package is configured to work properly with the `FLASH_TYPEERASE_PAGES` type erase.
[image]

Kind regards,

JayToltTech, 2024-09-30:
I think the confusion here is that the code in the samples refers to this as a sector erase but when you follow down into the logic it's actually performing page erases.

E.g. BLE_OTA flash_driver.c has functions named FD_EraseSingleSector, FD_EraseSectors, but these call into ProcessSingleFlashOperation which ends up using FLASH_TYPEERASE_PAGES.

closed 2024-11-19 by TOUNSTM