STMicroelectronics/STM32CubeWB #104: USBD_UnRegisterClassComposite always fails

x-moe-x, 2024-09-25:
The function `USBD_UnRegisterClassComposite(...)` in `Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_core.c` always fails.

Initially, `ret` is set to `USBD_FAIL` ...

https://github.com/STMicroelectronics/STM32CubeWB/blob/cbe89b3b4318fd8b1c8a1e51024fac568274c3d3/Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_core.c#L297-L304

... and on failure it is set again to `USBD_FAIL`

https://github.com/STMicroelectronics/STM32CubeWB/blob/cbe89b3b4318fd8b1c8a1e51024fac568274c3d3/Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_core.c#L316-L324

so on finish it fails  all the time.

https://github.com/STMicroelectronics/STM32CubeWB/blob/cbe89b3b4318fd8b1c8a1e51024fac568274c3d3/Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_core.c#L352-L360

ALABSTM, 2024-09-30:
Hi @x-moe-x,

Looks like you're right. Your report will be forwarded to our development teams. Thank you for the heads-up.

With regards,

ALABSTM, 2024-09-30:
ST Internal Reference: 192427

ALABSTM, 2024-10-31:
Hi @x-moe-x,

For your information, our development teams confirmed the issue. A fix will be available in a future release. We cannot share a date for the moment. I will keep you informed.

With regards,