STMicroelectronics/STM32CubeWB #109: Please rebuild stm32wb_zigbee_wb_lib.a with the latest compiler

sevaa, 2024-11-05:
The binary `stm32wb_zigbee_wb_lib.a` was built with IAR 9.20. That version of IAR had a problem with producing malformed `debug_aranges` section, which was acknowledged and fixed in IAR 9,30. As a result, binaries built against stm32wb_zigbee_wb_lib.a v1.20 make DWARF compliant tooling produce bogus results and/or crash.

Could you get the latest version of IAR (9.30+) and rebuild the library please?

RJMSTM, 2024-11-07:
Hello @sevaa,

This issue has already been fixed internally, and it will be available on the next release of STM32CubeWB v1.21.0 Firmware version.

Thanks,

sevaa, 2024-12-02:
I've got v1.21 updated No 28 - at least on the first file, pletoh.c, and on a couple of other files checked at random, the producer line still goes "IAR ANSI C/C++ Compiler V9.20.1.318/W64 for ARM", and the aranges section is still corrupt.

jiangshao6, 2024-12-08:
How did you learn to develop with this IAR?

sevaa, 2024-12-08:
Is that a question for me? I am not developing with IAR, I'm a user of the library. You can link your code to a static library with other toolchains.

RJMSTM, 2024-12-23:
ST Internal Reference: 199263

jiangshao6, 2025-01-04:
怎么使用freeRTOS?

sevaa, 2025-04-30:
Release 1.22 (2025-03-03) still compiled with IAR 9.20, aranges still malformed.

RJMSTM, 2025-07-02:
Hello,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.23.0 of the STM32CubeWB published recently on GitHub. Thank you again for having reported.

sevaa, 2025-07-02:
Confirmed, thank you. Compiler IAR v.9.60, and aranges looks right.