STMicroelectronics/STM32CubeWB #111: Micropython for Nucleo-68 WB55

mkphsnbng, 2025-03-27:
Hi,

I am trying to run micro python in Nucleo-68 STM32WB55R6VG.

I followed the link https://docs.micropython.org/en/latest/develop/gettingstarted.html#building-the-stm32-port

I am able to generate the binary files(hex,dfu,bin,elf)

I flashed firmware.bin file to 0x08000000.. and reset the board i am not able to see the REPL python on serial COM port(STLINK virtual COM port )

to verify wether serial port is configured correctly or not i print message in my project and able to observe log which we have printed.

but we are not able to observe the REPL in the same console.. (COM port)

to cross verify i have build mboot binary too and flashed in 0x08000000 and firmware.bin into 0x08000000

but still not able to get REPL prompt

 

Request to please help on this

RJMSTM, 2025-03-28:
Hello @mkphsnbng,

Thank you for your report. I will try to reproduce the problem. Otherwise, for further support, could you please post your issue on this repository [micropython](https://github.com/micropython/micropython) They might be able to assist you better with your problem.

Regards,

mkphsnbng, 2025-03-28:
Yes I have already posted!!! 😊
Micropython for Nucleo-68 WB55 · micropython · Discussion #17024<https://github.com/orgs/micropython/discussions/17024>

ALABSTM, 2025-04-25:
Hi @mkphsnbng,

I am afraid we cannot be of much help on this one. As indicated in the [CONTRIBUTING.md](https://github.com/STMicroelectronics/STM32CubeWB/blob/master/CONTRIBUTING.md#3-support-requests-and-questions) guide, I recommend that you submit your point to the [ST Community](https://community.st.com/t5/stm32-mcus-products/bd-p/stm32-mcu-products-forum), hoping you will find there the answers you are looking for.

With regards,